paradise-fi/RoFI #85: Update configuration

Ondraceq, 2021-09-30:
QoL improvements:
- [ ] Move to a more precise namespace (`rofi::configuration`)
- [ ] Make the header names more precise (e.g. use the `configuration/rofibot.hpp` pattern as in atoms)
- [ ] Use `ModuleHandle` for internal representation (which is not just a typedef) and export `ModuleId` for external usage
- [ ] Rename `Orientation` to be more specific, or move it to a nested namespace
- [ ] Move angle manipulating functions to `detail` namespace
- [ ] Rename `makeJoint` to be more precise (e.g. `makeComponentJoint`)
- [ ] I'm not sure how `std::vector< int >` exactly match joints in `Component`. Maybe use a typedef `JointId` or something
- [ ] `bodies` in `Module` can be written as `components().subspan( _connectorCount )`. (Similar for `connectors`.)
- [ ] In `component`, `body` and `connector`, I would like to have an `assert`
- [ ] I don't think that `Module::clone` should be there like this - at least make it return a `unique_ptr`
- [ ] The `Rofibot::getModule` method makes no sense (dereferencing and then taking the address). With that said, I feel like returning a reference would be better.

Ondraceq, 2021-09-30:
- [ ] Add a way to distinguish user programmable joints from inner joints (at least an iterator that returns the non-rigid joints if that is the right condition)

yaqwsx, 2021-10-01:
What's the use case for "Add a way to distinguish user programmable joints from inner joints"?

Ondraceq, 2021-10-03:
Whenever the user code wants to get a joint with a given number, I have to be able to distinguish what are the user programmable joints. (E.g. by `getJoint(2)` in rofi hal`.)

yaqwsx, 2021-10-03:
You can always tell that by the number of arguments. From the design perspective; all joints are "user-programmable". There are no hidden joints.

Ondraceq, 2021-11-02:
@jajik When adding dependences to `vtk`, check the `BUILD_HEADLESS` flag in cmake

jajik, 2021-11-02:
> @jajik When adding dependences to `vtk`, check the `BUILD_HEADLESS` flag in cmake

Should be fixed in the latest c7fb8d712c6ca7163da6221caf4235d541416bcb

closed 2022-01-17 by jajik